cp2004/OctoPrint-EEPROM-Marlin #70: [Bug]: After Marlin Update, ignore crc mismatch, restoring a config with bed mesh causes M112 ERROR.

skl111, 2023-02-11:
### Description of the bug

When updating marlin firmware and due to changes a CRC mismatch occurs, and you select ignore to load the default values, selecting restore a backup containing a mesh map will cause the printer to stop with and M112 Error.
If you run bed leveling or Bed visualizer before restoring, then printer will not stop with M112 error. 

### Plugin version

3.3.0

### OctoPrint version

1.8.6

### M503 Response

```gcode
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z800.00 E691.50
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P500.00 R500.00 T1000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 X5.00 Y5.00 Z0.30 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S1 Z10.00 ; Leveling ON
Recv: echo:  G29 W I0 J0 Z-0.08625
Recv: echo:  G29 W I1 J0 Z-0.03575
Recv: echo:  G29 W I2 J0 Z-0.00600
Recv: echo:  G29 W I3 J0 Z0.02025
Recv: echo:  G29 W I4 J0 Z0.02700
Recv: echo:  G29 W I5 J0 Z0.02875
Recv: echo:  G29 W I6 J0 Z-0.00225
Recv: echo:  G29 W I7 J0 Z-0.02600
Recv: echo:  G29 W I8 J0 Z-0.06225
Recv: echo:  G29 W I0 J1 Z-0.14425
Recv: echo:  G29 W I1 J1 Z-0.07100
Recv: echo:  G29 W I2 J1 Z-0.03375
Recv: echo:  G29 W I3 J1 Z-0.00600
Recv: echo:  G29 W I4 J1 Z0.00825
Recv: echo:  G29 W I5 J1 Z0.01025
Recv: echo:  G29 W I6 J1 Z-0.00775
Recv: echo:  G29 W I7 J1 Z-0.02700
Recv: echo:  G29 W I8 J1 Z-0.04775
Recv: echo:  G29 W I0 J2 Z-0.18075
Recv: echo:  G29 W I1 J2 Z-0.10825
Recv: echo:  G29 W I2 J2 Z-0.06950
Recv: echo:  G29 W I3 J2 Z-0.03200
Recv: echo:  G29 W I4 J2 Z-0.01450
Recv: echo:  G29 W I5 J2 Z-0.00325
Recv: echo:  G29 W I6 J2 Z-0.00750
Recv: echo:  G29 W I7 J2 Z-0.03200
Recv: echo:  G29 W I8 J2 Z-0.05625
Recv: echo:  G29 W I0 J3 Z-0.20675
Recv: echo:  G29 W I1 J3 Z-0.13050
Recv: echo:  G29 W I2 J3 Z-0.09225
Recv: echo:  G29 W I3 J3 Z-0.05050
Recv: echo:  G29 W I4 J3 Z-0.02675
Recv: echo:  G29 W I5 J3 Z-0.01575
Recv: echo:  G29 W I6 J3 Z-0.02575
Recv: echo:  G29 W I7 J3 Z-0.04200
Recv: echo:  G29 W I8 J3 Z-0.05325
Recv: echo:  G29 W I0 J4 Z-0.20475
Recv: echo:  G29 W I1 J4 Z-0.12975
Recv: echo:  G29 W I2 J4 Z-0.07975
Recv: echo:  G29 W I3 J4 Z-0.04025
Recv: echo:  G29 W I4 J4 Z-0.00825
Recv: echo:  G29 W I5 J4 Z0.00500
Recv: echo:  G29 W I6 J4 Z-0.00100
Recv: echo:  G29 W I7 J4 Z-0.01850
Recv: echo:  G29 W I8 J4 Z-0.03800
Recv: echo:  G29 W I0 J5 Z-0.23675
Recv: echo:  G29 W I1 J5 Z-0.15675
Recv: echo:  G29 W I2 J5 Z-0.10800
Recv: echo:  G29 W I3 J5 Z-0.06950
Recv: echo:  G29 W I4 J5 Z-0.04575
Recv: echo:  G29 W I5 J5 Z-0.03175
Recv: echo:  G29 W I6 J5 Z-0.03200
Recv: echo:  G29 W I7 J5 Z-0.05075
Recv: echo:  G29 W I8 J5 Z-0.07075
Recv: echo:  G29 W I0 J6 Z-0.21225
Recv: echo:  G29 W I1 J6 Z-0.13075
Recv: echo:  G29 W I2 J6 Z-0.08800
Recv: echo:  G29 W I3 J6 Z-0.05125
Recv: echo:  G29 W I4 J6 Z-0.02950
Recv: echo:  G29 W I5 J6 Z-0.01375
Recv: echo:  G29 W I6 J6 Z-0.02200
Recv: echo:  G29 W I7 J6 Z-0.04300
Recv: echo:  G29 W I8 J6 Z-0.07175
Recv: echo:  G29 W I0 J7 Z-0.17800
Recv: echo:  G29 W I1 J7 Z-0.09950
Recv: echo:  G29 W I2 J7 Z-0.05950
Recv: echo:  G29 W I3 J7 Z-0.02975
Recv: echo:  G29 W I4 J7 Z-0.00475
Recv: echo:  G29 W I5 J7 Z-0.00300
Recv: echo:  G29 W I6 J7 Z-0.01825
Recv: echo:  G29 W I7 J7 Z-0.03625
Recv: echo:  G29 W I8 J7 Z-0.06850
Recv: echo:  G29 W I0 J8 Z-0.08225
Recv: echo:  G29 W I1 J8 Z-0.03100
Recv: echo:  G29 W I2 J8 Z0.00300
Recv: echo:  G29 W I3 J8 Z0.03450
Recv: echo:  G29 W I4 J8 Z0.04625
Recv: echo:  G29 W I5 J8 Z0.04925
Recv: echo:  G29 W I6 J8 Z0.02925
Recv: echo:  G29 W I7 J8 Z0.01425
Recv: echo:  G29 W I8 J8 Z-0.03150
Recv: echo:; X-Twist Correction:
Recv:   M423 A0.00 I110.00
Recv:   M423 X0 Z0.00
Recv:   M423 X1 Z0.00
Recv:   M423 X2 Z0.00
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H220.00 B60.00 F255
Recv: echo:  M145 S1 H235.00 B75.00 F255
Recv: echo:; Hotend PID:
Recv: echo:  M301 P15.95 I1.30 D48.96
Recv: echo:; Bed PID:
Recv: echo:  M304 P235.81 I45.98 D806.24
Recv: echo:; User thermistors:
Recv: echo:  M305 P0 R4700.0 T94162.0 B4160.0 C0.000000000 ; HOTEND 0
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X29.50 Y-19.50 Z-2.10 ; (mm)
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X860 Y900 Z1000
Recv: echo:  M906 T0 E550
Recv: 
Recv: echo:; StallGuard threshold:
Recv: echo:  M914 X99 Y99
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:; Input Shaping:
Recv: echo:  M593 X F40.00 D0.15
Recv: echo:  M593 Y F40.00 D0.15
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L200.00 U100.00 ; (mm)
```

### API Response

```json
{"backups":[{"name":"eeprom_backup-20230210-211836","time":"2023-02-10 21:18:36"}],"eeprom":{"advanced":{"command":"M205","params":{"B":20000.0,"E":5.0,"J":null,"S":0.0,"T":0.0,"X":8.0,"Y":8.0,"Z":0.3}},"autolevel":{"command":"M420","params":{"S":false,"Z":10.0}},"bed_pid":{"command":"M304","params":{"D":497.3,"I":6.12,"P":110.38}},"delta":{"command":"M665","params":{"A":null,"B":null,"C":null,"H":null,"L":null,"R":null,"S":null,"X":null,"Y":null,"Z":null}},"endstop":{"command":"M666","params":{"X":null,"Y":null,"Z":null}},"feedrate":{"command":"M203","params":{"E":50.0,"X":500.0,"Y":500.0,"Z":10.0}},"filament":{"command":"M200","params":{"D":1.75}},"filament_change":{"command":"M603","params":{"L":200.0,"U":100.0}},"filament_runout":{"command":"M412","params":{"D":null,"H":null,"S":null}},"home_offset":{"command":"M206","params":{"X":0.0,"Y":0.0,"Z":0.0}},"hotend_mpc":{"command":"M306","params":{"A":null,"C":null,"F":null,"H":null,"P":null,"R":null}},"hotend_pid":{"command":"M301","params":{"D":48.96,"I":1.3,"P":15.95}},"input_shaping":{"command":"M593","params":{"D":null,"F":null}},"linear":{"command":"M900","params":{"K":0.0}},"material":{"command":"M145","params":{"B":null,"F":null,"H":null,"S0":{"B":60.0,"F":255.0,"H":220.0},"S1":{"B":75.0,"F":255.0,"H":235.0}}},"max_acceleration":{"command":"M201","params":{"E":5000.0,"X":500.0,"Y":500.0,"Z":100.0}},"print_acceleration":{"command":"M204","params":{"P":500.0,"R":500.0,"T":1000.0}},"probe_offset":{"command":"M851","params":{"X":29.5,"Y":-19.5,"Z":0.0}},"steps":{"command":"M92","params":{"E":691.5,"X":80.0,"Y":80.0,"Z":800.0}},"tmc_current":{"command":"M906","params":{"E":null,"T0":{"E":550.0,"X":null,"Y":null,"Z":null},"X":860.0,"Y":900.0,"Z":1000.0}},"tmc_hybrid":{"command":"M913","params":{"E":null,"X":null,"Y":null,"Z":null}}},"info":{"additional":{"EXTRUDER_COUNT":"1","FIRMWARE_NAME":"Marlin bugfix-2.1.x (Feb 10 2023 20:47:45)","MACHINE_TYPE":"SOVOL SV06","PROTOCOL_VERSION":"1.0","SOURCE_CODE_URL":"github.com/MarlinFirmware/Marlin","UUID":"cede2a2f-41a2-4748-9b12-c55c62f367ff"},"capabilities":{"ARCS":true,"AUTOLEVEL":true,"AUTOREPORT_POS":true,"AUTOREPORT_SD_STATUS":false,"AUTOREPORT_TEMP":true,"BABYSTEPPING":true,"BINARY_FILE_TRANSFER":false,"BUILD_PERCENT":false,"CASE_LIGHT_BRIGHTNESS":false,"CHAMBER_TEMPERATURE":false,"CONFIG_EXPORT":true,"COOLER_TEMPERATURE":false,"CUSTOM_FIRMWARE_UPLOAD":false,"EEPROM":true,"EMERGENCY_PARSER":true,"EXTENDED_M20":true,"HOST_ACTION_COMMANDS":true,"LEVELING_DATA":true,"LFN_WRITE":true,"LONG_FILENAME":true,"MEATPACK":true,"MOTION_MODES":false,"MULTI_VOLUME":false,"PRINT_JOB":true,"PROGRESS":false,"PROMPT_SUPPORT":false,"REPEAT":false,"RUNOUT":false,"SDCARD":true,"SD_WRITE":true,"SERIAL_XON_XOFF":false,"SOFTWARE_POWER":false,"THERMAL_PROTECTION":true,"TOGGLE_LIGHTS":false,"VOLUMETRIC":true,"Z_PROBE":true},"is_marlin":true,"locked":false,"name":"Marlin bugfix-2.1.x (Feb 10 2023 20:47:45)"},"stats":{"failed":0,"filament":"","finished":0,"longest":"","prints":0,"total_time":""}}
```

### OctoPrint logs

[octoprint (14).log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/10712441/octoprint.14.log)

cp2004, 2023-02-26:
Apologies for somehow missing this issue, I have only just picked it up.

> selecting restore a backup containing a mesh map

This plugin doesn't back up the saved mesh at the moment, it only backs up what you can configure in the UI. A plugin like [UBL Mesh Edit](https://plugins.octoprint.org/plugins/ublmeshedit/) can do that. If your backup has a mesh map in it, that is interesting - your log doesn't seem to show any commands like that being sent to the printer from the plugin to 'attempt' to restore the mesh either.

This is how I understand this issue:

* EEPROM is reset to default values - this doesn't include anything for the saved mesh, which is empty
* You restore the backup in the plugin, but this doesn't restore the mesh
* You go to print, but it doesn't start because the printer can't load a mesh (mesh saved slot is empty)

If you do it after running bed levelling, there is the extra step:

* EEPROM is reset to default values - this doesn't include anything for the saved mesh, which is empty
* You run bed levelling, which saves a mesh in EEPROM
* You restore the backup in the plugin, doesn't touch the mesh
* You go to print, it now has a mesh to use so the printer is happy?

If I've understood it correctly, this is expected behaviour with how these things were designed to work. Let me know if that's not right.

skl111, 2023-02-26:
Thanks for the reply, life is nuts. 
I updated the title as no bed mesh is stored in the backup, that was my mistake.
So this is what happens.

EEPROM is reset to default values - this doesn't include anything for the saved mesh, which is empty
You restore the backup in the plugin, but this doesn't restore the mesh
Printer Halts M112

EEPROM is reset to default values - this doesn't include anything for the saved mesh, which is empty
You run bed levelling, which saves a mesh in EEPROM
You restore the backup in the plugin, doesn't touch the mesh
Printer is happy no halt.

After looking at the process, maybe a check to see if a mesh is stored before restoring the config could be added to the plugin warning users it will halt the printer.

Using this on a lot of printers and when the marlin config gets a crc mismatch due to marlin update on them, none of which is the plugins fault, and forgetting to run the bed level when one is in a hurry can drive you nuts.