cp2004/OctoPrint-EEPROM-Marlin #73: [Request]: Using eprom locked on iPad

Mardak62, 2023-04-03:
### What do you need help with?

when connecting to octoprint web interface 

and trying to use eprom on an Ipad, I get the message:

Read Only

looks like the printer is locked

see attached screen capture

But no such issue when connecting from My Mac.

[image]

cp2004, 2023-04-03:
What version of iOS is your iPad running?

Mardak62, 2023-04-04:
I am using an old iPad with iOS 9.3.5 to follow the print when I am running around the building.

Thank you
brads
Amir

cp2004, 2023-04-04:
iOS 9.3.5 is too old to run this plugin in the browser unfortunately, since it uses some modern JavaScript features that are not available there.

Please also go to https://octoprint.org/browser-check - OctoPrint 1.9.0 modernises some of the JavaScript there too, and so if any of the tests there show 'no' you won't be able to use this iPad to access OctoPrint 1.9.0 in the future at all, it won't just be this plugin that is broken. This decision was made on the basis of only 0.15% of users (who opted into anonymous usage tracking) still running old browsers to connect to OctoPrint, we didn't want to hold back development any more.

Mardak62, 2023-04-04:
Understood, thank you very much. One last question, I need to fix my hotend xyz POS offset (installed a new swiss direct drive); Is that a possibility for the future versions do that  in eprom reader? or maybe it is possible and I am missing something. Thank you best regards

cp2004, 2023-04-04:
Sounds like you might need the 'Home offsets' part (M206), which will show up if your printer allows these to be configured.

cp2004, 2023-04-05:
I'm going to close this issue as the root problem is something I won't fix. Feel free to still add comments if necessary.

closed 2023-04-05 by cp2004