cp2004/OctoPrint-EEPROM-Marlin #77: [Bug]: Cannot install on octorint 1.10.1

JohnyPeaN, 2024-05-16:
### Description of the bug

Uprgraded octoprint from 1.10.0 to 1.10.1 and restored configuration. This plugin fails on installation.

<code>error: subprocess-exited-with-error
× python setup.py egg_info did not run successfully.
│ exit code: 1
╰─> [18 lines of output]
/tmp/pip-req-build-g2y9i4b0/versioneer.py:521: SyntaxWarning: invalid escape sequence '\s'
] = '''# -*- coding: utf-8 -*-
</code>

### Plugin version

3.3.0

### OctoPrint version

1.10.1

### M503 Response

```gcode
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E400.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E40.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X3000.00 Y2000.00 Z60.00 E10000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P1000.00 R1500.00 T1250.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 X12.00 Y12.00 Z0.60 E14.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S1 Z0.50 ; Leveling ON
Recv: echo:  G29 W I0 J0 Z-0.23250
Recv: echo:  G29 W I1 J0 Z-0.29000
Recv: echo:  G29 W I2 J0 Z-0.29000
Recv: echo:  G29 W I3 J0 Z-0.20000
Recv: echo:  G29 W I4 J0 Z-0.19250
Recv: echo:  G29 W I0 J1 Z-0.31500
Recv: echo:  G29 W I1 J1 Z-0.29250
Recv: echo:  G29 W I2 J1 Z-0.29250
Recv: echo:  G29 W I3 J1 Z-0.22250
Recv: echo:  G29 W I4 J1 Z-0.21500
Recv: echo:  G29 W I0 J2 Z-0.26500
Recv: echo:  G29 W I1 J2 Z-0.31000
Recv: echo:  G29 W I2 J2 Z-0.29750
Recv: echo:  G29 W I3 J2 Z-0.24500
Recv: echo:  G29 W I4 J2 Z-0.20250
Recv: echo:  G29 W I0 J3 Z-0.28750
Recv: echo:  G29 W I1 J3 Z-0.32500
Recv: echo:  G29 W I2 J3 Z-0.25500
Recv: echo:  G29 W I3 J3 Z-0.25250
Recv: echo:  G29 W I4 J3 Z-0.22250
Recv: echo:  G29 W I0 J4 Z-0.19750
Recv: echo:  G29 W I1 J4 Z-0.27250
Recv: echo:  G29 W I2 J4 Z-0.30000
Recv: echo:  G29 W I3 J4 Z-0.31500
Recv: echo:  G29 W I4 J4 Z-0.28750
Recv: echo:; Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:; Hotend PID:
[... 454 lines cut ...]

2024-05-16 10:39:17,823 - py.warnings - WARNING - /usr/lib/octoprint/lib/python3.12/site-packages/octoprint_ultimakerformatpackage/__init__.py:29: SyntaxWarning: invalid escape sequence '\.'
  self.regex_extension = re.compile("\.(?:gco(?:de)?(?:\.ufp)?)$")

2024-05-16 10:39:17,840 - py.warnings - WARNING - /usr/lib/octoprint/lib/python3.12/site-packages/octoprint_powerfailure/__init__.py:51: SyntaxWarning: invalid escape sequence '\s'
  self.MOVE_RE = re.compile("^G0\s+|^G1\s+")

2024-05-16 10:39:17,841 - py.warnings - WARNING - /usr/lib/octoprint/lib/python3.12/site-packages/octoprint_powerfailure/__init__.py:52: SyntaxWarning: invalid escape sequence '\s'
  self.X_COORD_RE = re.compile(".*\s+X([-]*\d*\.*\d*)")

2024-05-16 10:39:17,841 - py.warnings - WARNING - /usr/lib/octoprint/lib/python3.12/site-packages/octoprint_powerfailure/__init__.py:53: SyntaxWarning: invalid escape sequence '\s'
  self.Y_COORD_RE = re.compile(".*\s+Y([-]*\d*\.*\d*)")

2024-05-16 10:39:17,841 - py.warnings - WARNING - /usr/lib/octoprint/lib/python3.12/site-packages/octoprint_powerfailure/__init__.py:54: SyntaxWarning: invalid escape sequence '\s'
  self.E_COORD_RE = re.compile(".*\s+E([-]*\d*\.*\d*)")

2024-05-16 10:39:17,841 - py.warnings - WARNING - /usr/lib/octoprint/lib/python3.12/site-packages/octoprint_powerfailure/__init__.py:55: SyntaxWarning: invalid escape sequence '\s'
  self.SPEED_VAL_RE = re.compile(".*\s+F(\d*\.*\d*)")

2024-05-16 10:39:18,094 - octoprint.plugin.core - INFO - Plugin SSD1306-OLED-Display (0.1.4) is disabled.
2024-05-16 10:39:18,102 - py.warnings - WARNING - /usr/lib/octoprint/lib/python3.12/site-packages/octoprint_prusaslicerthumbnails/__init__.py:45: SyntaxWarning: invalid escape sequence '\.'
  self.regex_extension = re.compile("\.(?:gco(?:de)?|tft)$")

2024-05-16 10:39:21,037 - octoprint.plugins.pluginmanager - ERROR - Installing the plugin from https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/latest/download/release.zip failed, could not parse output from pip. See plugin_pluginmanager_console.log for generated output
2024-05-16 10:39:26,419 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.2.0.32

cp2004, 2024-05-16:
Unfortunately the plugin is not yet compatible with Python 3.12, I think none of my plugins are due to the same issue. I'll be able to fix it at some point in June, I hope.

automontronic, 2024-05-18:
I ran into the same issue. Looks like OctoPrint is no longer using versioneer (referenced in the error): https://github.com/OctoPrint/OctoPrint/commit/579bfccb2c1fcf07eb0c3cebcb2d5b18fa2e17da.

JohnyPeaN, 2024-06-11:
I have managed to install this plugin by replacing versioneer.py with git version from master of official versioneer (https://github.com/python-versioneer/python-versioneer), repacking back into zip and installing from file. Works for octolapse, too.

automontronic, 2024-07-22:
I have confirmed replacing versioneer.py works. Good find!

ZIOOIZ8, 2024-09-19:
I'm glad that it's working again for you. I installed Octoprint using a script from paukstelis/octoprint_deploy, which was probably packed into a venv. Unfortunately, I'm not very familiar with Linux/Ubuntu, so your suggested solution doesn't help me. Could you please describe in a little more detail what needs to be done to get it running again?

goonterknutso, 2024-10-02:
Is there any movement on this? I would like to use it but can't. Can someone make a branch with the fixed versioneer.py until an official release has it fixed? Or can you describe the steps to replacing the versioneer.py file?

goonterknutso, 2024-10-02:
Also, I see the devel branch has a fix in it. Is the devel branch able to be used for install so I can work around the python 3.12 issue for now? Thanks!

Getn67, 2024-10-29:
Are there any updates on moving the dev branch to production? The fix has been available since July. @cp2004

cp2004, 2024-10-29:
Sorry sorry sorry I know I said I'd do this. I'll try and get it out as soon as I can.

classicrocker883, 2025-02-28:
well seeing as foosel committed https://github.com/OctoPrint/OctoPrint/commit/579bfccb2c1fcf07eb0c3cebcb2d5b18fa2e17da on Oct 5, 2023, by "Replace versioneer with a custom tool"

then it should be just as easy to replace `versioneer` with that "custom tool"

here are those changes from that commit, but that was over 1.5 years ago so at least its a starting point at what changes can be made here

<br>

```diff
[‎MANIFEST.in]
- include versioneer.py

+include src/octoprint/_static_version.py

...

[setup.py]
- from distutils.command.build_py import build_py as _build_py

+ from setuptools.command.build_py import build_py as _build_py

...

[‎src/octoprint/plugins/softwareupdate/__init__.py]
- from octoprint._version import get_versions

+ from octoprint import __version__ as octoprint_version
```

kverkind, 2025-05-16:
> I have managed to install this plugin by replacing versioneer.py with git version from master of official versioneer (https://github.com/python-versioneer/python-versioneer), repacking back into zip and installing from file. Works for octolapse, too.

Can someone document the "how to ..." Thank you

classicrocker883, 2025-05-18:
> > I have managed to install this plugin by replacing versioneer.py with git version from master of official versioneer (https://github.com/python-versioneer/python-versioneer), repacking back into zip and installing from file. Works for octolapse, too.
> 
> Can someone document the "how to ..." Thank you

that's a pet peeve of mine, programmers never really explain "how to", as if they assume we *already know*. it's frustrating. some README's don't even tell what the thing is! it's like "here you go... figure it out"

anyway, here is basically what you need to do from the [README](https://github.com/python-versioneer/python-versioneer?tab=readme-ov-file#quick-install)

so what you actually have to do is:
1. Check this "How to use `pip` [here](https://packaging.python.org/en/latest/tutorials/installing-packages/)
2. basically just do this:
`python3 - m pip install -U e SomeProject @ git+https://git.repo/some_pkg.git`
3. or this, unpack and
`python3 - m pip install -U e <path>`

svn2208, 2025-09-14:
> I have managed to install this plugin by replacing versioneer.py with git version from master of official versioneer (https://github.com/python-versioneer/python-versioneer), repacking back into zip and installing from file. Works for octolapse, too.

I tried this approach and used AI to fix the code, since it seems to be only affecting versioning. For me it worked with: 
OctoPrint 1.11.3 Python 3.13.5 OctoPi* 1.1.0 (build 2025.09.09.113029) on a RPI4

Feel free to clone `https://github.com/svn2208/OctoPrint-EEPROM-Marlin.git` and `/opt/octopi/oprint/bin/pip3 install .`

the PR is here: #83