cp2004/OctoPrint-EEPROM-Marlin #82: [Bug]:  RE: Command M503 does not work on builds compiled from GitHub Actions

classicrocker883, 2025-02-19:
### Description of the bug

Related Issue: https://github.com/classicrocker883/MRiscoCProUI/issues/196

i'm not sure if this is a sort of duplicate of #81 or not,

but the issue has to do with Github Actions with `actions/**`

M503 does not work. for example:
the M503 command sends response 
````
Recv: echo:Unknown command: "M503"
- or -
Unknown command: "M503"
````
### Actions similar to around versions ~
- actions/cache@v4.2.0
- actions/setup-node@v4.1.0
- actions/checkout@v4.2.2
- actions/setup-python@v5.3.0
- - python-version: '3.11'
- bruceadams/get-release@v1.3.2
- tanyagray/action-upload-release-asset@main

At one point, back in October everything was fine and all the firmware files were created and uploaded successfully. Then in subsequent releases I updated the `actions/` versions and the `M503` command stopped working.

here is the file I currently am having issue with
- 2025-February [compile-configs.yml](https://github.com/classicrocker883/MRiscoCProUI/blob/2025-February/.github/workflows/compile-configs.yml)

here is the file that had work without issue
- 2024-October [compile-configs.yml](https://github.com/classicrocker883/MRiscoCProUI/blob/October-2024-Test/.github/workflows/compile-configs.yml)

<br>

---

For additional data and logs please let @natey416 comment and fill these in
for now, I am going to assume they are **up-to-date** versions

### Plugin version

TBA

### OctoPrint version

TBA

### M503 Response

```gcode
TBA

Recv: echo:Unknown command: "M503"
Unknown command: "M503"
```

### API Response

```json
TBA
```

### OctoPrint logs

TBA

cp2004, 2025-02-20:
Hey, thanks for getting in touch. The GitHub actions in this repository have nothing to do with the issue you're seeing. Running the M503 command against your firmware is completely independent of this plugin - you might as well have not installed it. There's nothing I can do to magically fix the firmware.

classicrocker883, 2025-02-28:
This issue has been since fixed. Thanks anyway for taking a look.
I'm wondering if you have any issues you need help with i may be able to contriibute.

closed 2025-02-28 by classicrocker883